GrumpyOldPizza/ArduinoCore-stm32l0 #95: Enabling Watchdog Timer

rishisharma920, 2019-08-02:
Hello, 

I am using the Grasshopper board for my testing and want to deploy a watchdog timer. Can you help me with the code snippet on how can I enable a WDT? For reference, I use the following when I need to restart my code on Arduino: wdt_enable(WDTO_2S);

GrumpyOldPizza, 2019-08-02:
STM32L0.wdtEnable(timeout)
STM32L0.wdtReset()

The "timeout" is in millisconds. Upper limit is 18000ms. Hope that helps.

rishisharma920, 2019-08-02:
Hello, 

Thank you for your prompt reply. I tested out the functions and they worked. But, I have another issue, I cannot disable the watchdog timer before going to stop mode to be in low power mode. Because of which the board resets with a maximum sleep time of 18 seconds.

GrumpyOldPizza, 2019-08-02:
Yep, that is correct. This is how the hardware works. See reference manual, 24.3.4.

rishisharma920, 2019-08-02:
Sure, but is there a way to disable it, once enabled?

GrumpyOldPizza, 2019-08-02:
No it cannot be disabled once enabled. That is kind of the whole purpose of a watchdog. If your system gets stuck by a software fault, there should not be a way that the software is able to disable the watchdog that could recover from such a fault.

closed 2020-04-26 by GrumpyOldPizza