GrumpyOldPizza/ArduinoCore-stm32l0 #103: How to attach interrupt to PA0?

BoelD, 2019-09-07:
I have a custom made board and use the Grasshopper board files. I want to use PA0 (WKUP) to wake up the device from sleep. I have no problem with waking up the device with a button attached to digital pin 1, but when I try the same code to use with PA0 (which is pin A4 in the definitions), it doesn't wake up after a pin change on WKUP.

How can I make the code below work for PA0?

`void setup()
{
  pinMode(1, INPUT);
  attachInterrupt(1, blinkLed, CHANGE);
}

void loop()
{
STM32L0.sleep();
}

void blinkLed()
{
  STM32L0.wakeup();
  digitalWrite(LED1, HIGH);
  delay(1000);
  digitalWrite(LED1, LOW);
}`

BoelD, 2019-09-07:
I solved this now by changing the first line of the` variant.cpp` file:

`    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA10), STM32L0_GPIO_PIN_PA10,             (PIN_ATTR_EXTI)`

into

`    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA0), STM32L0_GPIO_PIN_PA0,             (PIN_ATTR_EXTI)`

Since I am not using PA10 in my design. But I do wonder whether there is a way to do this without touching the board files?

GrumpyOldPizza, 2019-09-07:
Not sure which variant file you are working off, but Grasshopper uses A4 for PA0. And there it is declared like this (more or less):

`{ GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA0),  STM32L0_GPIO_PIN_PA0,   (PIN_ATTR_WKUP1 | PIN_ATTR_TAMP)
`

Your declaration will not work, because some internal use prevents PIN_ATTR_EXTI on PA0 (so either your interrupt will work, or LoRaWAN, but not both).

BoelD, 2019-09-07:
The interrupt worked, but I have not tested it with LoRaWan yet. So I will change it back.

I am working with the Grasshopper variant file. And let me rephrase the question: is it possible to attach an interrupt to PA0 when using this varaint file? This code does not work:

```
void setup() {
  pinMode(4, OUTPUT);
  pinMode(A4, INPUT);
  attachInterrupt(digitalPinToInterrupt(A4),flits,CHANGE);
  digitalWrite (4, HIGH);
  delay(500);
  digitalWrite (4, LOW);
}

void loop() {}

void flits()
{
  digitalWrite (4, HIGH);
  delay(500);
  digitalWrite (4, LOW);
}

```

BoelD, 2019-09-07:
I have it working now. I have PA0 attached to a HALL sensor, which is HIGH when there is no magnet around. When I use `FALLING` instead of `CHANGE` the interrupt works as intended.

Thanks for your help!

closed 2020-04-26 by GrumpyOldPizza