GrumpyOldPizza/ArduinoCore-stm32l0 #104: Strange thing with analogWrite

hallard, 2019-09-13:
I got 2 led connected to A3 and A2 as Follow

[image]

When I drive LED with DigitalWrite no issue but as soon as I use AnalogWrite got issue
Here my so complex code
```
void setup() 
{
}

void loop() 
{
  // Full Led Green ON
  analogWrite(LED_GRN, 0); 
  delay(500);
  // Full Led Green OFF
  analogWrite(LED_GRN, 255); 
  delay(1000);

  // Full Led Blue ON
  analogWrite(LED_BLU, 0); 
  delay(500);
  // Full Led Blue OFF
  analogWrite(LED_BLU, 255); 
  delay(1000);
}
````

When I set AnalogWrite to 255 to light off the Led the led does not go off, I mean it goes far lower light level but does not goes off, likes the PWM was not totally 100% High level

I tried adding in setup ` analogWriteResolution(255);` but still no changes

The only way I got it working is to set back pin to digital mode as follow

````
  analogWrite(LED_BLU, 0); 
  delay(500);
  pinMode(LED_BLU, OUTPUT);
  digitalWrite(LED_BLU, HIGH); 
  delay(1000);
````
Any Idea what I'm doing wrong ?

hallard, 2019-09-13:
Self Answer `analogWrite(LED_BLU, 256);` does the trick. So use 256 not 255 as Arduino.

GrumpyOldPizza, 2019-09-13:
Guess that is a bug. 0 and 255 (or whatever the boundaries) should turn the output to a constant 0 or 1.