GrumpyOldPizza/ArduinoCore-stm32l0 #106: Question: Is there a Capacitive Sense example for STM32L082XX?

applecrusher, 2019-10-03:
According to the STM32L082 documentation on page 28, [STM32L082 datasheet](https://www.st.com/resource/en/datasheet/stm32l082kb.pdf), the STM32L082 has the ability to do capacitive sensing. Is there an example of this in the repository? Other capacitive sensing libraries I have tried such as this [one here](https://www.arduinolibraries.info/libraries/capacitive-sensor), are not compatible and give me a complication error.

GrumpyOldPizza, 2019-10-04:
The TSC controller is not yet supported. At the end of the day, the available pins do not really lend themselfs to TSC capability.

applecrusher, 2019-10-04:
Thank you! If possible, can you elaborate on the second sentence? I have never done capacitive sensing and would greatly appreciate your input.

GrumpyOldPizza, 2019-10-04:
Please consult the documentation for the TSC peripheral and the availble TSC groups/pins on the Murata module ...

applecrusher, 2019-10-05:
Will do. Thank you.

closed 2019-10-05 by applecrusher