GrumpyOldPizza/ArduinoCore-stm32l0 #112: Strange Join Issues

hallard, 2019-11-27:
I've got multiple TTN Gateways (EU 868) around and noticed sometimes that join can take minutes.

This is really strange because sometimes it join quickly and sometimes not, could it be related to limited downlinks by TTN?

Anyway, and my question, is there any issue doing some ```STM32L0.stop(anyDelay)``` until we joined the network ? Does the stack still working async whatever we do in `loop()`

GrumpyOldPizza, 2019-11-27:
You might want to check with your gateway, whether it sees the join requests or not ... 

Regarding STM32L0.stop() and sich. Yes, the LoRaWAN stack works async in the background (also switching between RUN/SLEEP/STOP).

hallard, 2019-11-27:
Thanks for the explanation.
The join request goes to TTN I can see on TTN dashboard but that's all. I suspect it occurs when there is more than one GW sending to TTN. Need to do further investigation and also test on my own LoRaWAN server to see.

hallard, 2019-12-04:
I tested on my private LoRaWAN server and it works like a charm, after each reset of the board join is accepted immediately in a couple of seconds. So I tried back on TTN and tooks minutes.
So I decided to remove this line (found in some examples) that I don't know why it was for and only for TTN, but given a try
```
LoRaWAN.addChannel(1, 868300000, 0, 6);
```
Guess what, now it is joining immediately each time, so I think you can remove from examples.

GrumpyOldPizza, 2020-05-06:
Can I close this out now ?

hallard, 2020-05-06:
I could add your trick for 20ms error as a sample but may be best to wait this function in the framework so I can post an example here and close the issue?

GrumpyOldPizza, 2020-05-06:
Added a new API to the LoRaWAN class:

LoRaWAN.setSystemMaxRxError(millis);

The default is 10ms.

closed 2020-05-06 by GrumpyOldPizza