GrumpyOldPizza/ArduinoCore-stm32l0 #114: Build Error Issue

codezoo-ltd, 2019-12-02:
In file included from C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\libraries\Wire\src/Wire.h:23:0,

                 from C:\Users\jbmaster\Desktop\HumidityTemperature\HumidityTemperature.ino:10:

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h: In constructor 'Callback::Callback()':

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h:35:28: error: 'NULL' was not declared in this scope

     Callback() : _callback(NULL), _context(NULL) {  }

                            ^

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h: In constructor 'Callback::Callback(void (*)())':

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h:37:87: error: 'NULL' was not declared in this scope

     Callback(void (*function)(void)) : _callback((void (*)(void*))function), _context(NULL) { }

                                                                                       ^

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h: In member function 'Callback::operator bool()':

C:\Users\jbmaster\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10\cores\arduino/Callback.h:68:44: error: 'NULL' was not declared in this scope

     operator bool() { return (_callback != NULL); }

Hi, 
https://github.com/codezoo-ltd/sme-hts221-library
The above error occurred when I built it to work on Grasshopper LoRa Board.

So as a result of applying the source code as below
The library is working fine. Of course, the temperature and humidity results are good.

Can you upload a patch?

------ Code here ------ TleraCorp\hardware\stm32l0\0.0.10\cores\arduino\Callback.h

#pragma once

#define NULL 0    // Patch Here !!

class Callback {
public:
    Callback() : _callback(NULL), _context(NULL) {  }

    Callback(void (*function)(void)) : _callback((void (*)(void*))function), _context(NULL) { }

UT2UH, 2020-04-29:
I also had the issue. The better way can be to replace NULL with nullptr...  [callback.h](https://github.com/UT2UH/ArduinoCore-stm32l0/blob/K33RTC/cores/arduino/Callback.h)

GrumpyOldPizza, 2020-05-02:
Fixed.

closed 2020-05-02 by GrumpyOldPizza

kent-williams, 2020-05-12:
@GrumpyOldPizza Unfortunately I'm still experiencing the same issue here even with master.

When Arduino libraries outside of your core include "Wire.h", there is still no definition for the `NULL` as is still used in this [line](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/cd6186aec5440c057c7bbff6132f8a810a2bb4ac/cores/arduino/Callback.h#L33).

thanks!

waymond91, 2020-09-03:
So I'm compiling adafruits sensor library for the LIS3DH and am getting this error when I include their libraries (which in turn includes SPI & Wire).
How am I supposed to fix this?

kevin192291, 2021-03-07:
Hello All,
I just started playing with this for the grasshopper-L082CZ and it may be my new favorite dev board!
However, I am getting this very same error mentioned above. Has there been any progress made on fixing this?
If I need I can update the AdaFruit library.

Thanks everyone for this awesome library!!!