GrumpyOldPizza/ArduinoCore-stm32l0 #116: How to program using the SWD port

rmeldo, 2019-12-21:
Hi all ,
I hope you can point me in the right direction.
I finally have managed to solder the Murata module onto a board I designed, and managed to connect to it with a ST-Link dongle (I still don't have USB).
Now I would like to upload a sketch. In the Arduino IDE  I selected Sketch --> Export Compiled Library to export   a file to the upload using the ST-Link.
However the exported file is of the .dfu type, whereas the ST-Link utility needs a .hex file.
Do you have any advice on how to proceed?
Thanks
Riccardo

GrumpyOldPizza, 2019-12-22:
You can directly use a ST-Link device from the IDE ... So why do you want to go throu this extra step ?

If there is a good reason to do so, you need to change platform.txt to generate a hex file via objcopy.

rmeldo, 2019-12-22:
I tried to do it from the IDE but it doesn't seem to see the ST-Link and
doesn't program.

Any hints about changing the platform.txt?   It is beyond my capability.

Thanks
Riccardo

GrumpyOldPizza, 2019-12-22:
The IDE support works just fine. You need to select "ST-Link (STM32L0)" in "Tools -> Programmers", and then download the sketch via "Sketch -> Upload via Programmer".

If you do that, what error does the IDE throw ?

rmeldo, 2019-12-22:
Thanks,
a couple of errors

The first time I ran it I got the following:

GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00487-gaf359c18
(2018-05-12-15:41)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
embedded:startup.tcl:60: Error: Can't find
C:\Users\Riccardo\Documents\Arduino\hardware\TleraCorp\stm32l0/variants/Grasshopper-L082CZ/openocd_scripts/grasshopper-l082cz.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 60
the selected serial port at file "embedded:startup.tcl", line 60
 does not exist or your board is not connected

So I investigated why it couldn't find the file  "grasshopper-l082cz.cfg" .
In the directory I found  a file with a different name: "
grasshopper-082cz.cfg". So I changed its name and I got the error below.

Any suggestions?
Riccardo

GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00487-gaf359c18
(2018-05-12-15:41)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 300 kHz
adapter_nsrst_delay: 100
none separate
none separate
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x080000d4 msp: 0x20000838
STM32L0: Enabling HSI16
adapter speed: 1800 kHz
** Programming Started **
auto erase enabled
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000838
wrote 28672 bytes from file
C:\Users\Riccardo\AppData\Local\Temp\arduino_build_602901/stm32_serial.ino.elf
in 4.331460s (6.464 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20000838
verified 25992 bytes in 0.183826s (138.081 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 240 kHz
shutdown command invoked

GrumpyOldPizza, 2020-04-27:
Looks like the program via SWD was successful. Other than the bad filename, it did work ? Can I close this issue out then ?

closed 2020-05-02 by GrumpyOldPizza