GrumpyOldPizza/ArduinoCore-stm32l0 #117: TX Power not adjusted when antenna gain is changed

ThomasBaumgartner, 2019-12-22:
I have the following problem: I have a custom antenna on the grasshopper-node. I adjusted the antenna gain in the code with: 
LoRaWAN.setAntennaGain(1.2) after that i rechecked the TX-Power with: LoRaWAN.getTxPower(). I expected that the tx power is adjusted if the antenna gain is changed but nothing happened. I even tried with a bunch of different values for antenna gain but still the same tx-power.

GrumpyOldPizza, 2019-12-22:
TxPower is the logical transmit power. The gain is the gain/loss via the antenna path. Two different things.

closed 2019-12-22 by GrumpyOldPizza

ThomasBaumgartner, 2019-12-22:
Does this mean, that changing antenna gain does not influence the device in it's physical behaviour? 

In the LoRaMac.h file i found this calculation claiming that the tx power is calculated using max ERIP and antenna gain. Am I missing something? 
[image]