GrumpyOldPizza/ArduinoCore-stm32l0 #118: 61 days in December

UT2UH, 2020-01-02:
HNY and many thanks for the core!
RTC.getDay() preloaded from GPS gave me 61 days in December. 
I think array elements should have +31 in values starting from  stm32l0_rtc_days_since_month[x][14] up.

GrumpyOldPizza, 2020-01-02:
Can you elaborate more about the exact circumstances ?

UT2UH, 2020-01-02:
When I tried to reproduce the issue with this sketch [https://pastebin.com/WdhUPitr](url) I saw that  
`RTC.setDate( day, month, year); ` and 
`  RTC.setDay(day);
  RTC.setMonth(month);
  RTC.setYear(year);
`
gave me different output with the same initial date
`const byte day =   29;
const byte month = 12;
const byte year =  19;`

RTC.setDate() gave me an output 58.01.19, while RTC.setDay(day); RTC.setMonth(month); RTC.setYear(year); gave me ant output 28.01.20 - both wrong.
[https://github.com/UT2UH/RTC](url)

UT2UH, 2020-01-11:
I finally get back my Discovery board and made [a sketch to reproduce RTC lib issue](https://pastebin.com/qpfSdER4) - epoch time for year 2020 days is loaded to RTC first with setEpoch() , get back with getEpoch(), recalculated to calendar date using the code copypasted from stm32l0_rtc_time_to_calendar() and printed using getDay(), getMonth(), getYear(). 

Starting from 29/01/20, loaded and stored epoch time are different and getDay() and getMonth() print anything but valid date - [https://pastebin.com/eatXix6C](https://pastebin.com/eatXix6C). 
To fix the date recalculated with the code copypasted from stm32l0_rtc_time_to_calendar() I had to increment month indexes for stm32l0_rtc_days_since_month[][] table but the same changes in the core code still gave no correct results.

UT2UH, 2020-01-11:
[Yet another test sketch](https://pastebin.com/1zUC5ey0) shows [different stored epoch time](https://pastebin.com/BMmtHaKr) depending on the order of setDay(), setMonth(), setYear() calls... And there are no order giving the correct time.

GrumpyOldPizza, 2020-04-06:
Didn't get to this earlier. I just submitted some changes on github. Can you check whether the issue is fixed now ?

UT2UH, 2020-04-07:
Thank you very much, it is still important for my project! I will merge our changes and test them later today. BTW, I found that I forked and modified not the latest release but the master branch with unreleased patches. ) In the latest release available RTC worked much better but as far as I understand from the master branch patches there were also some issues.  Thank you for fixing them!

UT2UH, 2020-04-08:
It's a pity but I have tried exactly the same patch yet in January - it does not fix the issue - [rtc fix](https://github.com/UT2UH/ArduinoCore-stm32l0/commit/791773bd586eb92fb62e18684be8540aeeb7a200) I think that the problem is in leap year and leap year Feb offset in conversion tables but I can continue to debug only at the weekend.

GrumpyOldPizza, 2020-04-27:
Dopped some new code reworking the RTC handling. Mind checking out the latest on github for feedback whether the issue had been fixed now ?

UT2UH, 2020-04-29:
Sorry, can get the test board only today. Thank you for fixes! Now RTC works as expected. get() methods return the same values as set() in any order of D-M-Y and H:M. There is only 1s mismatch  between getEpoch and setEpoch in year 2021 - look at the end of [getEpoch](https://pastebin.com/nBxLWezV) Is it expected leap second change effect?

GrumpyOldPizza, 2020-04-29:
Good catch. I need to test this with 2021. The next leap second could be on
12/31/2020 23:59:59. So everything up to there uses hardcoded values. It
will get it wrong on the leap second (23:59:60). After that it should use
the value from GNSS or the default, which is 18, i.e. the same as before.
So a tad odd that you see this one second off.

UNIX epoch is define as seconds elapsed from 1/11/1970 in UTC time space
minus the leap seconds.  Yup. What a pisser. So not a fully linear time
scale.

UT2UH, 2020-04-29:
Sorry, I am not too good in the matter but here is the test sketch if you need one - [getEpochTest.ino](https://pastebin.com/7vnG7wwq).

GrumpyOldPizza, 2020-05-02:
Ok, leftover from debugging. Fixed now.

UT2UH, 2020-05-02:
Thank you for the fixes!

closed 2020-05-02 by UT2UH