GrumpyOldPizza/ArduinoCore-stm32l0 #119: Unable to build with TinyGSM library

hallard, 2020-01-09:
I'm trying to build basic sample `AT_Debug` with [TinyGSM](https://github.com/vshymanskyy/TinyGSM) library  but I can't build any of the example with error on `overriding 'virtual size_t Stream::read(uint8_t*, size_t) `

It works fine on ESP8266 or other boards so could it be a compiler option to setup?

Error is as follow

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/charles/Library/Arduino15/packages -hardware /Users/charles/OneDrive/devt/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/charles/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/charles/OneDrive/devt/Arduino/libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X1209_0X6667 -ide-version=10810 -build-path /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_build_54629 -warnings=default -build-cache /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_cache_876699 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM/tools/AT_Debug/AT_Debug.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/charles/Library/Arduino15/packages -hardware /Users/charles/OneDrive/devt/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/charles/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/charles/OneDrive/devt/Arduino/libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X1209_0X6667 -ide-version=10810 -build-path /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_build_54629 -warnings=default -build-cache /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_cache_876699 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM/tools/AT_Debug/AT_Debug.ino
Using board 'Cricket-L082CZ' from platform in folder: /Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10
Using core 'arduino' from platform in folder: /Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10
Detecting libraries used...
/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10810 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0 -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff "-DUSB_MANUFACTURER=\"Tlera Corporation\"" "-DUSB_PRODUCT=\"Cricket\"" -DUSB_TYPE=USB_TYPE_CDC -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Device/ST/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/cores/arduino -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/variants/Cricket-L082CZ /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_build_54629/sketch/AT_Debug.ino.cpp -o /dev/null
Alternatives for TinyGsmClient.h: [TinyGSM@0.9.17]
ResolveLibrary(TinyGsmClient.h)
  -> candidates: [TinyGSM@0.9.17]
/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10810 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0 -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff "-DUSB_MANUFACTURER=\"Tlera Corporation\"" "-DUSB_PRODUCT=\"Cricket\"" -DUSB_TYPE=USB_TYPE_CDC -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Device/ST/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/cores/arduino -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/variants/Cricket-L082CZ -I/Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM/src /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_build_54629/sketch/AT_Debug.ino.cpp -o /dev/null
Alternatives for STM32L0.h: [STM32L0@1.0]
ResolveLibrary(STM32L0.h)
  -> candidates: [STM32L0@1.0]
/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10810 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0 -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff "-DUSB_MANUFACTURER=\"Tlera Corporation\"" "-DUSB_PRODUCT=\"Cricket\"" -DUSB_TYPE=USB_TYPE_CDC -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Device/ST/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/cores/arduino -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/variants/Cricket-L082CZ -I/Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM/src -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/STM32L0/src /var/folders/9z/bvhxpn7d6d907h0sqplvpp100000gp/T/arduino_build_54629/sketch/AT_Debug.ino.cpp -o /dev/null
/Users/charles/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10810 -DARDUINO_STM32L0_CRICKET -DARDUINO_ARCH_STM32L0 -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6664 -DUSB_DID=0xffff "-DUSB_MANUFACTURER=\"Tlera Corporation\"" "-DUSB_PRODUCT=\"Cricket\"" -DUSB_TYPE=USB_TYPE_CDC -DDOSFS_SDCARD=0 -DDOSFS_SFLASH=0 -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/CMSIS/Device/ST/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/system/STM32L0xx/Include -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/cores/arduino -I/Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10 [... cut ...]
 Used: /Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM
Multiple libraries were found for "STM32L0.h"
 Used: /Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/STM32L0
Using library TinyGSM at version 0.9.17 in folder: /Users/charles/OneDrive/devt/Arduino/libraries/TinyGSM 
Using library STM32L0 at version 1.0 in folder: /Users/charles/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/STM32L0 
exit status 1
Error compiling for board Cricket-L082CZ.
```

Thanks

GrumpyOldPizza, 2020-05-02:
Need to rethink that.

In a nutshell, Client.h defines "read()" always returning an "int" instead of a proper "size_t". Udp.h messes up the types there as well to offer "int read(char *, size_t)" and a "int read(unsigned char *, size_t)".

All the other logic in this core is using "size_t read(uint8_t *, size_t)".

GrumpyOldPizza, 2020-05-02:
This should be fixed now. Mind quickly checking via TOT github and closing out if fixed.

hallard, 2020-05-02:
Yeah, thanks, switching from different environment each time needed to change back this on will same me time :)

closed 2020-05-02 by hallard