GrumpyOldPizza/ArduinoCore-stm32l0 #120: LoRaWAN MAC-Test failed

ThomasBaumgartner, 2020-01-18:
After putting the device in test mode with the command: 01010101 on port 224 the device started sending in a 5s interval - so far so good. In a next step I send a 4 byte value starting with 0x04 (sort of Ping Pong test)- expecting the device to answer with a 4-Byte payload starting with 0x04 and adding 0x01 to each of the remaining bytes. But that did not happen, the device continued sending in it's 5s intervall but ignoring my downlink.
Am I missing something? Any help is highly appreciated.

GrumpyOldPizza, 2020-01-21:
Good question. That should actually work, as some folks use the compliance test mode to validate their devices.

Is a link check test done before that ?

ThomasBaumgartner, 2020-01-31:
Link check functionality is active using (in setup):
LoRaWAN.setLinkCheckLimit(100)
LoRaWAN.setLinkCheckDelay(3)
LoRaWAN.setLinkCHeckThreshold(2)
But directly before changing to the testmode the device did not send a link check. Can link check affect the behaviour in testmode?

ThomasBaumgartner, 2020-05-14:
I was finally able to successfully pass the LoRaWAN-MAC Test. Disabling almost the entire application did it. I'm not 100% sure but I guess disabling the Watch-Dog did it.

closed 2020-05-14 by ThomasBaumgartner

aessig, 2020-11-23:
@ThomasBaumgartner Would you be able to share your code? We are struggling to run the compliance test?

ThomasBaumgartner, 2020-11-25:
#include "LoRaWAN.h"

const char *appEui  = "XXX123";
const char *appKey  = "**********************";
const char *devEui =  "XYZ123";

void setup( void )
{
    LoRaWAN.begin(EU868);
    LoRaWAN.addChannel(1, 868300000, 0, 6);
    LoRaWAN.setADR(true);
    LoRaWAN.setDataRate(5);
    LoRaWAN.setDutyCycle(true);
    LoRaWAN.joinOTAA(appEui, appKey, devEui);
}

void loop( void )
{
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        LoRaWAN.beginPacket();
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.write(0xad);
        LoRaWAN.write(0xde);
        LoRaWAN.endPacket();
    }

    delay(15000);
}

With this setup I was able to perform Compliance Test. I think setting Data Rate to 5 is not important, but since the device is sending every 15s it helps to fullfill dutycycle restrictions. Do you guys have testinfrastructure? Since we don't live far from each other, I could give you assistance.

aessig, 2020-12-08:
@ThomasBaumgartner Thank you very much for your message. It seems we have been able to make it work. That would be great, we don't have the test infrastructure yet. Should we discuss by email (aurelien.essig@oratek.ch)?