GrumpyOldPizza/ArduinoCore-stm32l0 #123: Sending data in lora packets from usart 2 (B L072 LRWAN)

meakash, 2020-03-18:
For my application, i need to transmit data in lora packets from usart 2. I declared serial 2 in hardware serial library folders but they seem to work only when the on board reset button is pressed.

Can you please help to resolve this issue and how to transmit this data through lorawan 

I have already used usart 1 for some other sesnor.

GrumpyOldPizza, 2020-03-18:
Again, not sure what you are trying to do. The B-L072-LRWAN variant already offers 2 UARTs:

/*
 * Serial interfaces
 */

#define SERIAL_INTERFACES_COUNT 2

#define PIN_SERIAL_RX        (0ul)
#define PIN_SERIAL_TX        (1ul)

#define PIN_SERIAL1_RX       (2ul)
#define PIN_SERIAL1_TX       (8ul)

There is nothing you need to change. So if you changed something, all bets are off.

meakash, 2020-03-18:
[image]

According to this image D0 and D1 are used as serial pins and D2 and D8 are used as serial 1 pins.

[image]

Similarly i have to define PA3 and PA2 as Serial 2 
please help me to do the following

GrumpyOldPizza, 2020-03-18:
What you describe is exactly what the code does as is.

"Serial" maps to D0/D1 (PA3/PA2)

"Serial1" maps to D2/D8 (PA10/PA9).

I am not sure what you are trying to do that has not been done yet. Please use UM2115, table 5 as reference.

closed 2020-04-26 by GrumpyOldPizza