GrumpyOldPizza/ArduinoCore-stm32l0 #127: STM32L0.stop(mywalue) sometime never wake

hallard, 2020-04-24:
Hi Thomas,

Today I have a board in production that I suspect to stop waking after `STM32L0.stop(10)` on extremely rare case.

To be honest it's not the 1st time I'm suspecting this problem and until today I thought this was impossible,  but it's very difficult to catch because the issue is very rare.
I'm seeing this on different application using LoRaWAN stack. one use an I2C sensor on other a DS18B20 so the case is relatively independant of hardware. 
I just had a call this my engeneer that developed the complex algorithm of one of this device to submit the issue regarding algorithm errors and he told me (and his answer makes sense) Are you sure you're not "blocked" somewhere into loop, you wait 10ms then do calculation then wait and so other on loop, so if you have no value for 15 min (Lora send every 15min) that's because I don't calculate them so you're stuck somewhere in loop (ho ho not waking?! strange I already saw that)

That's my issue, now I think I got the cause, I'm using timerMillis() to fire an event every 15 minutes to send LoRa Frame. The send is working perfectly but I got no values in frame, and I have them when all is working file. Seems to confirm alway in stop mode (and as all LoRa is async it continue to works)

But has I got push button on board with a callback from attatchinterrupt, once I push the button all is going fine as usual and start back to calculate values and not block anymore and I can see new values in the next LoRa frame.

Now looks like the puzzle is the async mode. Because all that is async continue to works as expected. And what I'm doing with timermillis(), is just to set his callback to the `sendMessage` of LoRa. See below

now that means the sendMessage is also executed async (when STM32 was in stop mode from loop for example) and I'm wondering if what I'm doing in this callback can have side effect on the "no wake" of the main loop ? For example the sendMessage callback read battery, VCC, other stuff ?

I will try to handle things different, I mean timerMillis to call a basic callback that only set flag and call `STM32L0.wakeup()`that will have the effect to wake up from loop and then test flag from loop and call `sendMessage()` in sync from main loop instead of timerMillis() calllback.

Any hints on what could cause this no wake to happens (Or I'm totally wrong?)

Thanks

Code example of what is done with timerMillis and sendMessage
```
void joinCallback(void)
{
  uint32_t nextTransmit = 60 ; // in 60s

  // Not Joined ?
  if (!LoRaWAN.joined()) {
    // This will force a rejoin in 2 min
    nextTransmit =  120 ;
  }

  debug("joinCallback() ");
  if (!LoRaWAN.joined()) {
    debug("NOT ");
  }
  debug("JOINED, next transmit in ");
  debug(nextTransmit);
  debugln(" seconds");
  transmitTimer.start(sendMessage, nextTransmit * 1000);
}

void doneCallback(void)
{
  uint32_t nextTransmit = config.lora_transmit ;

  // Classic data message
  fPort = FPORT_DATA;

  if (!LoRaWAN.linkGateways())  {
    // This will force a rejoin in 2 min
    nextTransmit = 120 ; 
  }

  if (received_config) {
    // Send config confirmation in 60s
    nextTransmit = 60 ; 
  } 

  debug("doneCallback() ");
  if (!LoRaWAN.linkGateways())  {
    debugln("DISCONNECTED ");
  }
  debug("next transmit in ");
  debug(nextTransmit);
  debugln(" seconds");

  transmitTimer.start(sendMessage, nextTransmit * 1000);
}

void sendMessage(void)
{
  float VCC = readVcc(); 
  float CoreTemp = STM32L0.getTemperature();
  float VBAT, VEXT; 

  readBatteries(&VBAT, &VEXT);

  transmitTimer.stop();

  debug("VCC:"); debug(VCC); 
  debug("  VBat:"); debug(VBAT); 
  debug("  VExt:"); debugln(VEXT);
  debug("MCU Temp: "); debug2ln(CoreTemp, 2);

  if (!LoRaWAN.busy())  {
   
    if (!LoRaWAN.linkGateways() || !LoRaWAN.joined()) {
      debugln("REJOIN()");
      LoRaWAN.rejoinOTAA();
    }
    
    if (LoRaWAN.joined()) {
      int16_t value;
      int8_t  vext = 0;

      debug("TRANSMIT( ");
      debug("fPort:");
      debug(fPort);
      debug(", TimeOnAir:");
      debug(LoRaWAN.getTimeOnAir());
      debug(", NextTxTime:");
      debug(LoRaWAN.getNextTxTime());
      debug(", DR:");
      debug(LoRaWAN.getDataRate());
      debug(", TxPower:");
      debug2(LoRaWAN.getTxPower(), 1);
      debug("dbm, UpLinkCounter:");
      debug(LoRaWAN.getUpLinkCounter());
      debug(", DownLinkCounter:");
      debug(LoRaWAN.getDownLinkCounter());
      debugln(" )");

      if (VUSB) {
        LoRaWAN.setBatteryLevel( LORAWAN_BATTERY_LEVEL_EXTERNAL );
      } else {
        LoRaWAN.setBatteryLevel( map(VBAT, 2.5f, 3.6f, LORAWAN_BATTERY_LEVEL_EMPTY, LORAWAN_BATTERY_LEVEL_FULL));
      }

       LoRaWAN.beginPacket(fPort);

      // Bat from 0 (0V) tp 250 (5V) 1 byte
      value = (uint8_t) (VBAT * 100 / 2) ;
      LoRaWAN.write(value);

          // CPU Temperature
          LoRaWAN.write( (int8_t) CoreTemp );

          // Maximum 10s counter
          LoRaWAN.write(max_pulse_10s >> 8);
          LoRaWAN.write(max_pulse_10s & 0xFF);

          // Send counter
          LoRaWAN.write(pulse_count >> 24);
          LoRaWAN.write(pulse_count >> 16);
          LoRaWAN.write(pulse_count >> 8);
          LoRaWAN.write(pulse_count & 0xFF);

          // Send differential counter
          LoRaWAN.write(pulse_transmit >> 24);
          LoRaWAN.write(pulse_transmit >> 16);
          LoRaWAN.write(pulse_transmit >> 8);
          LoRaWAN.write(pulse_transmit & 0xFF);

          // i2c errors
          LoRaWAN.write(i2c_error >> 8);
          LoRaWAN.write(i2c_error & 0xFF);

        // Clear differential counter and maximum pulse count
        pulse_transmit = 0;
        max_pulse_10s = 0;
        min_pulse_10s = 0xFFFF;
        i2c_error = 0;

      LoRaWAN.endPacket();
    }
  }
}
```

GrumpyOldPizza, 2020-04-24:
Not sure what could be going on. I kind of stopped reading by the time I saw DS18B20, which essentially requires interrupts to be disabled for a longer period of time. That would kill all the RTC/TimerMillis/LoRaWAN stack right there.

hallard, 2020-04-24:
Humm, Interesting, 

It happens also without DS18B20 just with one I2C device (magnetometer). I mean is there any restriction on what we can do in the callback of timerMillis() ? I mean I don't know if it's IRQ based, if it's the case maybe the callback should do less and as fast as possible? just an idea.

For DS18B20 I changed firmware 3 days ago to read the DS devices values in loop instead of callback of timermillis() (that was also sending lora) so it should happens only when LoRa stack is doing nothing, I'm sending after reading. Is it possible if I disable interrupts not to break stacks? If so how?

2nd point, when loop is in stop() and the timermillis() callback is fired, of course loop stays is stop() but can we use all peripherals (I2C, SPI, ..) as usual in the callback of timerMillis()?

GrumpyOldPizza, 2020-04-26:
Can you please retry with the latest code from github ?

hallard, 2020-04-27:
For sure I will do that, in the meantime I reworked the code on my side also, so not sure if it works where is the working fix, may be one, may be other, may be both. Anyway if it's working it will be fine.

GrumpyOldPizza, 2020-04-27:
I was going over the code again. Couple of questions:

(1) What do the functions readBatteries() and readVCC() do exactly ?

(2) The "10" milliseconds timeout is the issue I am worried about. If a
timeout is to small, then the RTC peripheral might choke on it. That why I
reworking the RTC code internally. So that issue might be gone now.

(3) STM32L0.stop() only wakes up otherwise if you tell it to via
STM32L0.wakeup(). So executing a callback will not bail out of
STM32L0.stop() ...

hallard, 2020-04-27:
These function are nothing fancy, here below the code, we got 2 batteries, one internal 3.6V and one external 12V

```cpp
float readVcc() 
{
  return STM32L0.getVDDA(); 
}

float readBatteries(float * main, float * ext) {

  uint16_t raw;
  float _vcc = STM32L0.getVDDA(); 

  pinMode(BAT_ENABLE, OUTPUT);
  digitalWrite(BAT_ENABLE, HIGH);
  delay(5);
  if (main) {
    raw = analogRead(BAT_ANALOG);
    *main = ((100.0f + 27.0f ) / 100.0f ) * _vcc * ((float) raw) / 4096.0f;
  }

  if (ext) {
    raw = analogRead(BAT_ANALOG_EXT);
    *ext  = ((100.0f + 470.0f ) / 100.0f ) * _vcc * ((float) raw) / 4096.0f;
  }

  digitalWrite(BAT_ENABLE, LOW);
  pinMode(BAT_ENABLE, INPUT);

  if (main) return *main;
  if (ext) return *ext;
  return 0;
}
```

Ok I'll try new code today, we plan to deploy tomorrow on customer boards.

Thank you very much for your help

GrumpyOldPizza, 2020-04-29:
Is this issue fixed with the latest drop ?

hallard, 2020-05-02:
Not sure, it's rarely, but I close it and will re open if needed

closed 2020-05-02 by hallard