GrumpyOldPizza/ArduinoCore-stm32l0 #137: Switching between LoRaWAN and FSK

petinox, 2020-06-02:
I have a few questions about how the radio is set up, I would be grateful for any help or advice.

I want the device to function as a node on LoRaWAN and then be able to switch the radio to FSK for other communication, and then back to LoRaWAN again. Is switching between LoRaWAN and FSK modes is supported by the API?

Right now I seem to have gotten it to work by disabling the check in LoRaWAN.begin() where _Band is checked to see if LoRaWAN is already initialized, and then running the init sequence from the LoRaWAN examples again. If I don't make that change, I can see the LoRa packet being sent on the air after switching LoRaWAN->FSK->LoRaWAN, but the packet is not received by the gateway. I suspect there is some radio settings that linger from the FSK mode. Is there a correct way to do this switch within the API? Or could such support be added easily?

Also, what radio settings are set in preparation for each transmit in LoRaWAN mode, and which are assumed to have been set during initialization? I have tried to look through the code to see what is going on, but it is not obvious to me.

SloMusti, 2020-06-03:
I have very briefly tested this and it appears that LoRaWAN.end() is missing to fully clean-up and reset the radio. Speculating a workaround could be to manually issue a reset to the radio.

closed 2020-06-06 by GrumpyOldPizza

petinox, 2020-08-03:
For anyone else messing around with this although the official support is lacking, I got the switching to work by doing the following:

1) In LoRaWAN.cpp, LoRaWANClass::begin(), comment out the following lines that checks if LoRaWAN is already initialized:

```
 if (_Band) {
        return 0;
    }
```
2) Before switching radio mode (either direction):

```

  while(FskRadio.busy()) {}
  
  while(LoRaWAN.busy()) {}
  
  FskRadio.end();  
```
And then setup FskRadio/LoRaWAN as usual. 

Due to the smooth session saving of the LoRaWAN library this works well for me so far.