GrumpyOldPizza/ArduinoCore-stm32l0 #140: Removal of XON/XOFF handling

UT2UH, 2020-06-03:
Can the removal of XON/XOFF handling break Serial connectivity on Discovery B-L072Z-LRWAN1?

GrumpyOldPizza, 2020-06-03:
Unlikely, but always possible. What exactly does not work, and do you have a small repro sketch to show the problem ?

UT2UH, 2020-06-03:
There was a simple RadioHead library example with three Serial.println in setup() and two of them are mostly incomplete at speeds higher than 9600. The board is Discovery B-L072Z-LRWAN1. At the same time my getEpoch  RTC test sketch prints at higher speed without any errors but do it burst by burst.  Another board is custom one I used with ST-Link v3 lost Serial connectivity completely - it can be my another fault but I can load firmware to it with updated OpenOCD. Both boards have ST-Link connected to USART1.

GrumpyOldPizza, 2020-06-03:
Ok, let's try this again. A simple sample sketch on B-L072Z-LRWAN1 that
shows the problem. No RadioHead, no ST-Link/V3. Just the simple standard
B-L072Z-LRWAN1 via the standard embedded ST-Link setup.

GrumpyOldPizza, 2020-06-06:
Retested. No issue there.

closed 2020-06-06 by GrumpyOldPizza