GrumpyOldPizza/ArduinoCore-stm32l0 #143: Serial not working?

silardg, 2020-06-08:
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/c11545bc9751a05f1b1e0a6b1f8130bcb0fb26dc/variants/B-L072Z-LRWAN1/variant.h#L105

I'm trying to get Serial to work on PA2, TX.
By the code above, PIN_SERIAL_TX is arduino pin 1, and Serial is setup by extern const stm32l0_uart_params_t g_SerialParams.

I tried the same pin with PlatformIO and it worked with Serial, pin 1 TX (PA2). 
The code I'm using is simple Serial.begin(9600); and Serial.println("test");

I'm using B-L072Z-LRWAN1.

What am I doing wrong?!

matthijskooijman, 2020-11-04:
IIUC, the "Serial" object refers to USB virtual (CDC) serial. The hardware UARTs are mapped to `Serial1` and upwards by default. Maybe you should be using `Serial1` instead?

GrumpyOldPizza, 2020-11-04:
Arduino sets this up kind of oddly in general.

On a real AVR based platform (other the ones with builtin USB) "Serial" refers to both, D0/D1 and the attached USB. This is what the code for B-L072Z-LWAN1 does. 

With other non-NUCLEO platforms, USB maps to "Serial", and hence D0/D1 map to "Serial1".

matthijskooijman, 2020-11-04:
Ah, I didn't realize that this board does not have native USB, I see, so then my suggestion does not apply and something else is going on.

tejeshw, 2021-01-30:
How to use Serial2 on PA2 and PA3?   i tried the HardwareSerial command . got error in compiler!