GrumpyOldPizza/ArduinoCore-stm32l0 #150: Question about using u-blox AssistNow with the GNSS library

petinox, 2020-08-01:
I would like to use AssistNow with the u-blox GPS on the Tlera Corp Gnat board. Is there a way to do this while playing nice with the GNSS library? Is it possible to do it as simple as Serial1.begin(19200); Serial1.write(<UBX Commands>); Serial1.end()?

There seems to be some coupling of the GNSS library with the power pins of the GPS as well, so I suppose doing this while GNSS is not initialized won't work as there might be a reset on GNSS.begin()?

Any pointers or experiences are very welcome.

SloMusti, 2020-08-08:
Using the default GNSS example should give you a good starting point and that works well with this library.

petinox, 2020-08-08:
Yes, the GNSS example works well, but it does not have any facility to send custom ublox commands to the GPS.

I have GPS assist data that lets me get a much faster GPS fix thus reducing the battery consumption, but I don't know how to use it in this setting. I need to send this data to to the GPS without disrupting the GNSS library communications.

I have just gotten remote swd debugging to work however, so I have high hopes that I'll be able to make some modifications and get this working. Since some of the boards targeted by this core are explicitly meant for GPS tracking I suspect this is of interest for other users as well, as this would reduce the battery consumption for GPS fixes considerably.

GrumpyOldPizza, 2020-08-08:
Sorry, AssistNow is only supported in Autonomous mode. Question really is "how do you get a huge chunk of AssistNow data via LoRWAN onto your end nodes". The max time window is about 4 weeks. So if you plug in your nose via USB it can only be in the field for 4 weeks and have this data.

petinox, 2020-08-08:
Absolutely true.

But in my application I am using FSK to transfer the data. I can rely on that there is a large probability that the node will be close to a FSK beacon at least once every four weeks, and probably more often than that. So I just ping for a beacon occasionally, which is cheap in terms of battery (a few ms of airtime), and ask the beacon for Assist data if available (a few seconds of airtime, mostly in RX). Not terribly complicated. Then I use LoRaWAN to report positions and status while away from the beacons.

I'm only missing that last step, serving the data to the GPS module. Then I have GPS fixes that are a factor 3-10 cheaper in terms of battery for the validity of the assist data. If the validity of the assist data runs out, then there is more battery left for the more expensive fixes. This is especially useful when GPS reception is poor, as it can be with mobile nodes.

hallard, 2020-09-03:
I'm in the same case. We have some boards that are doing HTTP request once a day to push GPS point to our backend (using a modem of course). As we are connected to send data, we use it to get back AssistNow data from uBlox API. And same as you need to think on how we can send this data to UBlox GPS since we do not have "high level" access to API.
Should not be too complicated the hardiest part is from my point of view to eat the UBlox datasheet protocol and data transfer