GrumpyOldPizza/ArduinoCore-stm32l0 #151: Sending float (sensor data) to Lora server

Nasserkhaled, 2020-08-08:
Dear GrumpyOldPizza,

I'm trying to transmit float datatype to the server but I'm getting error "call of overloaded 'write(float&)' is ambiguous".
It works with int but I really want to transmit the data as a float because the fractions in the reading are important for my application. Can you please help me?

petinox, 2020-08-08:
I suggest that you create a byte buffer from the float, and then convert the bytes back to a float on the receiving end. Here is a nice guide to working with byte buffers:

https://www.thethingsnetwork.org/docs/devices/bytes.html

Another alternative is to use the Cayenne LPP protocol. Check out the Cayenne LPP example for the Cayenne LPP library in this repo.

It is also much easier to get help if you post your code, at least the line that generates the error.

Nasserkhaled, 2020-08-10:
Thank you Mr Petinox for your answer
I have tried to find examples for the binary buffer but I just don't know how to use it. I'm not good with arduino.

Here is my code:

#include "LoRaWAN.h"
#include "MQ7.h"

MQ7 mq7(A0,5.0);

const char *appEui = "0101010101010101";
const char *appKey = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx";
const char *devEui = "xxxxxxxxxxxxxxxx";
float codensity = 0 ;

void setup( void )
{
    Serial.begin(9600);
    
    while (!Serial) { }

    LoRaWAN.begin(AS923);
    // LoRaWAN.setSubBand(2);
    // LoRaWAN.setDutyCycle(false);
    // LoRaWAN.setAntennaGain(2.0);
    LoRaWAN.joinOTAA(appEui, appKey, devEui);

    Serial.println("JOIN( )");
}

void loop( void )
{

    codensity = mq7.getPPM() ;
    Serial.println(codensity);
   // delay(1000);

    
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        Serial.print("TRANSMIT( ");
        Serial.print("TimeOnAir: ");
        Serial.print(LoRaWAN.getTimeOnAir());
        Serial.print(", NextTxTime: ");
        Serial.print(LoRaWAN.getNextTxTime());
        Serial.print(", MaxPayloadSize: ");
        Serial.print(LoRaWAN.getMaxPayloadSize());
        Serial.print(", DR: ");
        Serial.print(LoRaWAN.getDataRate());
        Serial.print(", TxPower: ");
        Serial.print(LoRaWAN.getTxPower(), 1);
        Serial.print("dbm, UpLinkCounter: ");
        Serial.print(LoRaWAN.getUpLinkCounter());
        Serial.print(", DownLinkCounter: ");
        Serial.print(LoRaWAN.getDownLinkCounter());
        Serial.println(" )");

        LoRaWAN.beginPacket();
        LoRaWAN.write(codensity);
        LoRaWAN.endPacket();
    }

    delay(3000);
}

The error : call of overloaded 'write(float&)' is ambiguous

GrumpyOldPizza, 2020-08-10:
The API is setup to exchange data buffers, not values. Hence write() is meant to write a series of bytes.

What you could do is this:

LoRaWAN.write((uint8_t*)&codensity, sizeof(codensity));

Nasserkhaled, 2020-08-10:
Thank you so so much Mr. GrumpyOldPizza and Mr. petinox
Both solutions worked with me 

Thank you

closed 2020-08-10 by Nasserkhaled