GrumpyOldPizza/ArduinoCore-stm32l0 #156: Unclear if it is possible to lock position fixes to 2D with GNSS

nedhorning, 2020-08-27:
I would like to lock the GNSS receiver into 2D mode and specify a fixed elevation using a variable. This seems to be possible with the MAX M8Q that I'm using (on the Gnat) but I couldn’t figure out how to set it up using the GNSS library or even if it is possible. Locking position fixes to 2D will likely improve horizontal position accuracy since I am working in a sea level environment. Any guidance would be appreciated.

Ned

GrumpyOldPizza, 2020-08-28:
That is really not supported.

Not sure how to do that to begin with. Perhaps setting dynModel to 5 in UBX-CFG-NAV5 and then force 2D fixes only ?

nedhorning, 2020-08-28:
I'm not sure either. I see there is an opMode setting that would have to be set to Manual and then set fixMode to 2D only, fixedAlt to the elevation and fixedAltVar for altitude variance (I'm not familiar with that). Having a 2D mode option would be nice but it's not critical.