GrumpyOldPizza/ArduinoCore-stm32l0 #158: Linux 32bit (Raspberry Pi Compute Module 3+)

aessig, 2020-09-08:
While trying to compile the code from an RPI, using arduino-cli, I have encountered the following error: 

`Error during build: fork/exec /root/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/tools/linux/dfu-suffix: exec format error`

From this issue #17 , it seems that the build tools have been changed for the from 32 to 64 bits.

How could we make it work for both ?

Best regards

mcradit, 2020-09-08:
Install dfu-util. Replace the 3 dfu-* files in the directory you mentioned above with the 3 that get install with dfu-util. I can't remember what directory they get installed in.

GrumpyOldPizza, 2020-09-08:
I would think there are 2 separate things here. One is that all the build tools have been move to 64 bit, for the Intel processors. That makes sense, because the 64 bit CPUs are now common for 15 years, and who would use a 32 bit OS anymore ...

The next item is that Raspberry Pi is a ARM CPU, not a Intel CPU. And that is simply not supported that at all.

aessig, 2020-09-16:
Thanks, @mcradit. It worked perfectly.

matthijskooijman, 2020-11-04:
One way to fix this is to not include precompiled binaries in the core / git repo itself, but use external "tool" zip files which are declared in the platform json file, and are installed by the IDE automatically. This allows using a different zip file for each architecture and thus supporting more different tools.

This is also the method used to install gcc, which refers to the arduino-supplied gcc tool, so that already works for different architectures. It seems that Arduino also supplies a dfu-util tool (also for ARM), which could be used by this core as well (unless maybe that version is too old?).

GrumpyOldPizza, 2020-11-04:
On my list of things to do. GCC will be retrieved from somewhere else. 

dfu-utils, I am not sure about yet, because we use our own version with some WARs for the ST USB/DFU variant. But, yes same issue. It was just easier to add that to the core archieve initially coming from L4.

Same goes for openocd perhaps as well. 

This would solve the 32bit/64bit Linux X86 issues, and the Linux ARM issues.

One big problem though is testing (for me).

baarratia, 2022-02-18:
@aessig Hi! I'm trying to do the same (compile the code from an RPI) and I've already replaced the 3 dfu-* files in the directory but now the Arduino IDE throws another error:

> /root/Arduino/hardware/TleraCorp/stm32l0/tools/linux/stm32flash: 2: Syntax error: "(" unexpected
> An error occurred while uploading the sketch

Did you had that same problem? If so, how did you fixed it? 

Thanks!