GrumpyOldPizza/ArduinoCore-stm32l0 #162: TimeMillis doesn't seem work properly with 2 timers

Sestre, 2020-10-21:
Hello,

First, thanks for this useful work!

I have an issue with the TimerMillis library when I used 2 timers at the same time and restart those timers with different timing. Here a minimal sketch who shows the error:
```
#include <STM32L0.h>
#include "LoRaWAN.h"
#include "TimerMillis.h"

#define INT_FLAG_1 0x01
#define INT_FLAG_2 0x02

TimerMillis timer1, timer2;
uint32_t t1 = 60 * 1000;  // 1min
uint32_t t2 = 360 * 60 * 1000;  // 360min
uint8_t n1 = 3;
uint8_t n2 = 3;
uint8_t interruptFlags = 0x00;

void int_handler1 () {
  Serial.println("int_handler1");

  __disable_irq();
  interruptFlags |= INT_FLAG_1;
  __enable_irq();
}

void int_handler2 () {
  Serial.println("int_handler2");

  __disable_irq();
  interruptFlags |= INT_FLAG_2;
  __enable_irq();
}

void setup()
{
  // ------ SERIAL ------
  Serial.begin(9600);
  delay(3000);

  // ------ TIMERS ------
  timer1.start(int_handler1, 10000, 10000); // Set on 10sec
  timer2.start(int_handler2, 16000, 16000); // Set on 16sec
  Serial.println("Timers started");
}

void loop()
{
  if(interruptFlags & INT_FLAG_1) {
    __disable_irq();
    interruptFlags &= ~INT_FLAG_1;
    __enable_irq();
    if(n1) handler1();
  }

  if(interruptFlags & INT_FLAG_2) {
    __disable_irq();
    interruptFlags &= ~INT_FLAG_2;
    __enable_irq();
    if(n2) handler2();
  }
}

void handler1() {
  if (n1 == 1) {
    // After 3 interrupts of timer1
    timer1.restart(t1, t1); // t1 = 1min
    Serial.println("Timer 1 restarted");
  }

  n1--;
}

void handler2 () {
  if(n2 == 1) {
    // After 3 interrupts of timer2
    timer2.restart(t2, t2); // t2 = 360min
    Serial.println("Timer 2 restarted");
  }

  n2--;
}
```

And here the Serial output:
```
15:11:03.065 -> Timers started
15:11:13.077 -> int_handler1
15:11:19.072 -> int_handler2
15:11:23.076 -> int_handler1
15:11:33.052 -> int_handler1
15:11:33.052 -> Timer 1 restarted
15:11:35.074 -> int_handler2
15:11:51.076 -> int_handler2
15:11:51.076 -> Timer 2 restarted
15:12:01.059 -> int_handler1 // should have arrived 1 min after the restart and not just 8sec after
15:13:33.059 -> int_handler1 // now it all good. The timer has "lost his lead"
15:14:33.051 -> int_handler1
15:15:33.057 -> int_handler1
...
```

I tried to replace "restart" with "stop" and "start" functions but it doesn't solve the issue.

If I set "n2" to "5" in place of "3", everything works perfectly. The issue seems to occur when "timer2.restart()" runs after "timer1.restart()" but before the first interrupt of timer1 (with his new time values) occurs.

Moreover, the issue doesn't seem to come from the Arduino Serial timestamp.

Have someone an idea of what I can do to solve that or what I'm doing wrong?

Thanks

nelisse, 2020-11-27:
We are also experiencing sometimes irregular timer behavior when using multiple timers. We had numerous code reviews, but nobody can pinpoint the cause. In our case the timer callback is also sometimes called to early, however it is hard to find a reproducible situation. In our timer callbacks we only set a global flag and wake up the processor (flag is polled and handled from loop()). Recently we found a reproducible situation that after an external interrupt occurs, the timer callback is called too early. This suggests that it is maybe not only timer handling related. When the external interrupts do no occur, timer behavior returns to normal....

GrumpyOldPizza, 2020-11-27:
Is this on the downloadable version, or the latest code directly off GitHub ?

nelisse, 2020-11-27:
Release 0.0.10 via Arduino Board Manager.
Which is/looks identical to 0.0.10 from GitHub Releases [(https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/releases)](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/releases)

GrumpyOldPizza, 2020-11-27:
I meant the sources off GitHub directly.

nelisse, 2020-11-29:
The current GitHub state is incompatible on an application level with the 0.0.10 release, so I have not been doing much testing with the current GitHub state.

I will need to have some way in my applications, to be able to distinguish between the 2 code lines (therefore my other question regarding the version literals).

Furthermore the Arduino IDE/CLI unfortunately does not have an easy way to switch between various package releases, so it requires some careful administration on my side….

I will come back to this later with updates!

 

Martin

hallard, 2020-12-09:
I already faced out this, here what I've done (on mac os so may be you need to adjust some folder names) to switch between version. 
Assuming you installed the repo from arduino IDE the package on MacOs are under
`~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10`

Then I got on my home dir both version named `0.0.10-git`and `0.0.10-release` and a script to switch from one to other

```shell
charles@mac-office:stm32l0$ pwd
/Users/charles/Devt/stm32l0
charles@mac-office:stm32l0$ ls -lA
total 24
-rw-r--r--@  1 charles  staff  10244 13 jul 14:27 .DS_Store
drwxr-xr-x  15 charles  staff    480  4 mai  2020 0.0.10-git
drwxrwxr-x@ 14 charles  staff    448 13 jul 11:24 0.0.10-release
lrwxr-xr-x   1 charles  staff     39  4 jui  2020 set_stm.sh -> /Users/charles/OneDrive/devt/set_stm.sh
charles@mac-office:stm32l0$ 
```

The trick is to link in the `packages` folder the git version or release version, it's done by a basic script

```shell
#/bin/bash

echo "Select environement for STM32L0"
echo "1) latest release"
echo "2) git"
echo -n "Input Selection : "
read version

if [ "$version" = "1" ]; then
        folder="release/"
elif [ "$version" = "2" ]; then
        folder="git/"
else
        echo "You have entered an invalid selection!"
fi

echo "Setting folder to $folder"

if [ "$folder" != "" ]; then
	echo "unlink ~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10"
	unlink ~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10
	echo "linking ~/devt/stm32l0/0.0.10-$folder to ~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10"
	ln -sFf ~/devt/stm32l0/0.0.10-$folder ~/Library/Arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10
fi

```

Hope this helps