GrumpyOldPizza/ArduinoCore-stm32l0 #168: Reason for no PIN_ATTR_EXTI on PA0 with LoRaWAN

gitMichael, 2020-11-06:
What's the exact reason for that? Mentioned at:

Your declaration will not work, because some internal use prevents PIN_ATTR_EXTI on PA0 (so either your interrupt will work, or LoRaWAN, but not both).

_Originally posted by @GrumpyOldPizza in https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/103#issuecomment-529108501_

GrumpyOldPizza, 2020-11-06:
PA0 aliases with PC0. Latter one is used for SX1276. So PA0 cannot have a EXTI on it, because that one is taken by PC0.

PA0 is instead routed as RTC TAMP pin, so same functionality ... almost.

gitMichael, 2020-11-07:
OK I see now. But I think you mean PB0 which is used for the SX1276 DIO2 IRQ input on the EXTI0 line (at least at cmwx1zzabz). Thanks.