GrumpyOldPizza/ArduinoCore-stm32l0 #174: B-L072Z-LRWAN1 Power Consumption

bob1033, 2021-02-24:
I've have been trying to lower the power consumption of the B-L072Z-LRWAN1 dev board I plan on using for a low power project. I have been having problems getting the consumption down to the low uA range that I've heard some people achieve. I am currently only able to get to 350uA during stop using the STM32L0.deepsleep() function.

I've made the following modifications to the board:

- Opened SB6 to disconnect regulators. I feed 3.3V from my power supply to the 3.3V pin and measure current with a shunt.
- Opened SB37 to release the reset line from the ST-Link
- JP9 is in the 1-2 position, so PA12 can control the TCXO

I'm not sure what could by causing the 350uA current. I'm hoping the more experienced may have a place to start looking?

I ran out of headers, so I haven't measured the current of the individual IDD current lines. I have more on order and can provide those measurements if they will help troubleshooting.

I'm using the current master branch of the library

A snippet of my main loop:
```C
void loop( void )
{
    char dataPayload[24];
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        Serial.print("TRANSMIT( ");
        Serial.print("TimeOnAir: ");
        Serial.print(LoRaWAN.getTimeOnAir());
        Serial.print(", NextTxTime: ");
        Serial.print(LoRaWAN.getNextTxTime());
        Serial.print(", MaxPayloadSize: ");
        Serial.print(LoRaWAN.getMaxPayloadSize());
        Serial.print(", DR: ");
        Serial.print(LoRaWAN.getDataRate());
        Serial.print(", TxPower: ");
        Serial.print(LoRaWAN.getTxPower(), 1);
        Serial.print("dbm, UpLinkCounter: ");
        Serial.print(LoRaWAN.getUpLinkCounter());
        Serial.print(", DownLinkCounter: ");
        Serial.print(LoRaWAN.getDownLinkCounter());
        Serial.println(" )");

        // Send Packet
        LoRaWAN.sendPacket((uint8_t *)payload, sizeof(payload), false);

    }
    Serial.end();       
    STM32L0.deepsleep(10000); //10 Seconds
    Serial.begin(9600);
}
```

GrumpyOldPizza, 2021-02-25:
Sorry, no clue. The ST DISCOVERY/NUCLEO platforms are not really that great
for low power prototyping (i.e. somewhat complicated).

You do not need to do Serial.end() / Serial.begin(), unless Serial points
to USB.

I suppose the ST-Link part could cause some of the consumption. LEDs are
always in the 200uA to 300uA range. I did check that the TCXO via PA12 is
hooked up correctly in the variant.cpp file.

What else is connected to the system ?

gitMichael, 2021-02-25:
It's absolute no problem to get Idd down to spec values with the B-L072Z-LRWAN1. That's it make for.

~340uA sounds like the DGB-Interface. Try open CN8 jumper and restart the mcu.
Or a running UART but it should be a LPUART and also not run in sleepsleep / stop mode with your setup.

bob1033, 2021-02-25:
Okay, some more testing:

- All LED solder bridges have been opened
- SB28 and SB29 opened which disconnects the Virtual com from the module
- CN8 is opened
- In my initial test, nothing except for the usb to the dev board was connected to the board, result is 350uA
- Somehow, with the USB removed, board now has draw of 425uA
 
This leads me to believe something with the onboard ST-link is causing the current draw. However, I don't know where the draw is occurring as the ST-link should be completely isolated from the module now.

@gitMichael Have you been able to get the spec values on the B-L072Z-LRWAN1 before?

gitMichael, 2021-02-25:
Yes, you can reach with ArduinoCore-stm32l0:
1.05uA for Imcu JP3 (SB14 open), 
0.15uA for Irf JP1 (SB17 open) and 
~0.1uA Iusb JP2 (SB19 open).

With only one deepsleep() function call in standard config with everything connected and initialized but SB37 open (nRST costs 4uA) it's possible w/o problems.

Try measuring currents at JP1..3 first.

bob1033, 2021-02-25:
Thanks Michael, I'll measure JP1-3 and report back.

bob1033, 2021-04-12:
I have figured out the issue causing the excessive current draw. In the schematic below the ST-Link is connected to the microcontroller's 3.3V rail on pin PA0 via a 4.7K resistor. This resistance isn't quite high enough and causes the rail to back-power the ST-Link through what I assume are some protection diodes. This causes the 400uA of excess current.

[image]

I fixed this issue by removing resistor R26 from the board. Now I'm reading a sleep current of around 2uA (including a low Q power supply I'm using).

closed 2021-04-12 by bob1033