GrumpyOldPizza/ArduinoCore-stm32l0 #180: Feature Request: RPC update a variable

hpssjellis, 2021-04-17:
@GrumpyOldPizza thank you so much for making this repo. I have been using the new [Arduino Portenta](https://store.arduino.cc/usa/portenta-h7) with [LoRa Vision Shield](https://store.arduino.cc/usa/portenta-vision-shield-lora) and your library is the only LoRa and LoRaWan library that is allowing me any success.  (Arduino types have made a few changes to get it working [here](https://github.com/facchinm/ArduinoCore-stm32l0) ) 

I have a method to exchange data between the 2 cores (Main M7 and inner M4) of the Portenta [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/my02c-dual-core-RPC-variable.ino)  but feel I need some sort of RPC availability on the Murata module running the LoRa chip. Do you have any suggestions?

GrumpyOldPizza, 2021-04-17:
No idea what to suggest there. This ArduinoCore is meant to be running a sketch on the STM32L0 on the muRata module. WHat you are asking is to run an application sketch on the Portenta (who came up with that name ...) ... So in reality you might as well install some AT-Command based firmware there and be done. Kind of a waste of power and resources, as it would have been equally easy just to put a SX1262 module onto the LoRa VIsion Shield and let CM4 handle the LoRaWAN stack.

hpssjellis, 2021-04-17:
LOL, I agree.  Lots of Portenta (yes about the name) hardware decisions I have issues  with: 

1. put an accelerometer on the shields, 

2: the HD connector is reversible so us makers have a 50% chance of messing up, 

3: seriously I could go on.    

 I have been asking for a basic AT command ability (to the MuRata module) for a while now, possibly no one I am asking at Arduino knows how to implement it (like me). 

Anyway, your method works, I don't really care about efficiency, I just like working product.  "I teach High School, up to the kids to make something real." I just want to send and perhaps receive a string from the murata module. Any suggestions, words to google? I couldn't even find a library for RPC on the STM32IO. Can the murata send I2C or SPI? I guess the hardware pins would need to be  connected. Could a string be sent along TX and RX like how the module is programmed?

Thanks for replying.

kevin192291, 2021-04-17:
Hey, just wanted to pipe up and say that both i2c and serial communication works great. I even had it wired up to a modbus board with no issues. Haven't tried spi yet but it would be great to have an sd card on board!

GrumpyOldPizza, 2021-04-17:
A quick thought. If you have the ST Discovery, there is a code somewhere, which you then can use to register on my.murata.com. There is a variant of the module that has an AT-Command set, which is on this web-site along with the manual. Perhaps it's possible to flash this onto the module. That would be the easiest way.

ST also has in their I-CUBE-LRWAN some AT-Command set. But that needs to be compiled first.

A better way would be to use something more along HCI (coming from BLE), and use SPI instead of UART (less power). But then again, that would be a lot of work.

hpssjellis, 2021-04-17:
Thanks @kevin192291 @GrumpyOldPizza.

 Here is some UART code I wrote for the Portenta to read data from the other UART's (I ws testing the new Breakoput board).   Would it be something similar on the murata module side that I just write to UART and see if the Portenta sees it? I believe the Portenta has a pre-defined serial term ```SerialLora``` which I should be able to insert in my code below: Can someone show me what code I would need on the murata side just to send "Hello World". I think if I can do one direction I should be able to figure out both directions.

```

#include "mbed.h"
#include "rtos.h"

using namespace mbed;  // sometimes needed
using namespace rtos;

Thread thread;

void myLedBlue_thread(){
   while (true) {
      digitalWrite(LEDB, !digitalRead(LEDB));   //switch on / off
      ThisThread::sleep_for(1000);
      Serial.println("Waiting...");
   }
}

void setup() {
   pinMode(LEDB, OUTPUT);   // LEDB = blue, LEDG or LED_BUILTIN = green, LEDR = red 

   Serial.begin(115200);
   Serial1.begin(115200);
   Serial2.begin(115200);
   Serial3.begin(115200);
   
   thread.start(myLedBlue_thread);
}

void loop() {

  if (Serial.available()) {     // If anything comes in Serial 
    Serial.write(Serial.read());   // read it and send it out Serial (USB)
  }
  if (Serial1.available()) {     // If anything comes in Serial1 
    Serial.write(Serial1.read());   // read it and send it out Serial (USB)
  }
  
  if (Serial2.available()) {     // If anything comes in Serial2 
    Serial.write(Serial2.read());   // read it and send it out Serial (USB)
  }
  
  if (Serial3.available()) {     // If anything comes in Serial3 
    Serial.write(Serial3.read());   // read it and send it out Serial (USB)
  }
  
}

```

hpssjellis, 2021-04-17:
@GrumpyOldPizza @kevin192291 The following compiles on the Portenta so I am reasonably happy. I think on the murata side serial UART is called STM_TX1 and STM_RX1

Just had an epiphany. The way the Portenta Programs this library is all serial, so I can just copy how it works. I Should be good now.

Here is what I have so far, but will probably change it.

```
#include <Arduino.h>
#include "mbed.h"
#include "rtos.h"

//using namespace mbed;  // sometimes needed
using namespace rtos;

int myLastUart = -1;
Thread thread;

UART mySerial8(PA_2,  PA_3,  NC, NC);  // murata modem,   TX, RX, RTS, CTS  NOTE: NC means not connected

void myLedBlue_thread(){
   while (true) {
      digitalWrite(LEDB, !digitalRead(LEDB));   //switch on / off
      ThisThread::sleep_for(1000);
      if (myLastUart >=0) {
         Serial.println("Last Serial message was from UART:" + String(myLastUart));
         myLastUart = -1;
        }
      Serial.println("Waiting...");
   }
}

void setup(){
    pinMode(LEDB, OUTPUT);   // LEDB = blue, LEDG or LED_BUILTIN = green, LEDR = red 
    Serial.begin(115200);
    mySerial8.begin(9600);   
    thread.start(myLedBlue_thread);
}

void loop(){
   if (mySerial8.available()) {    
      Serial.write(mySerial8.read()); 
      myLastUart = 8;  
   }
}
```

hpssjellis, 2021-04-18:
@GrumpyOldPizza @kevin192291 

Sorry for the long posts. I have a question: How do I activate UART on the murata module? This code fails. 
```
UART mySerialSTM32(STM_TX1,  STM_RX1,  NC, NC);  // murata modem,   TX, RX, RTS, CTS 
```

Better question, do you have a link for the STM32IO API?

hpssjellis, 2021-04-19:
@GrumpyOldPizza @kevin192291 

OK, well that solution was really easy.  Portenta UART3 reads STM32LO serial.    I think for the moment I am doing fine. 

This is all I need to get the Portenta reading any regular serial output from the murata chip. 
```
UART mySerial3(PJ_8,  PJ_9,  NC, NC);

  if (mySerial3.available()) {    
     Serial.write(mySerial3.read()); 
  }

```

I think I will close this issue.

P.S. Thomas: If I tidy up the Arduino code for the Portenta LoRa version of your library with a few examples specific to the Portenta, would you be interested in a PR from me for you to merge with your repository?

closed 2021-04-19 by hpssjellis