GrumpyOldPizza/ArduinoCore-stm32l0 #185: support for Visual-studio-code

gereic, 2021-04-28:
Is it possible to use the stm32l0 (Murata-module) with sx1262 also in Visual-studio-code ?
If yess, is there a how-to ?

kevin192291, 2021-04-28:
Hello @gereic !
I am currently using VS Code to write stuff for the murata module with Platform.io
You can get it in the extensions area. Once you have done that, click new project, and type "grass" and the select box will auto reduce to the Grasshopper-L082CZ.
Check this page out: https://docs.platformio.org/en/latest/boards/ststm32/grasshopper_l082cz.html
It shows the supported board. It is basically the same as the arduino IDE in terms of flashing, but you can create folders and have the type ahead of VS Code :)

gereic, 2021-04-28:
Very cool. I will check this out.
Do you know also this board ?
https://wireless.murata.com/type-1sj.html
Can this board also programmed with VS-Code ?

kevin192291, 2021-04-28:
I do not know, though it seems to have the same processor so possibly? I think both chips however are equally as difficult to get your hands on right now :(
Also I do not believe that the type1-sj is telec certified incase you were looking for that :/

GrumpyOldPizza, 2021-04-28:
> Hello @gereic !
> I am currently using VS Code to write stuff for the murata module with Platform.io
> You can get it in the extensions area. Once you have done that, click new project, and type "grass" [...]

That platform-io configuration does not use this ArduinoCore. Not sure that is what you are after.

Have you tried the Arduino IDE 2.0 beta ? That is I think VS Code based as well.

GrumpyOldPizza, 2021-04-28:
> Very cool. I will check this out.
> Do you know also this board ?
> https://wireless.murata.com/type-1sj.html
> Can this board also programmed with VS-Code ?

Like that one here ?

[image]

gereic, 2021-04-28:
That is what I'm searching for, where can I buy this module ?

GrumpyOldPizza, 2021-04-28:
> That is what I'm searching for, where can I buy this module ?

You cannot buy this, yet. It's a prototype. Will take a few weeks/months.

kevin192291, 2021-04-28:
@GrumpyOldPizza 
> That platform-io configuration does not use this ArduinoCore. Not sure that is what you are after.

Ohhh Sorry for the incorrect info!!! I had thought I was using arduinocore (oops!), what is this using???

gereic, 2021-05-09:
Is the Murata 1SJ-Module already supported on Arduino ?
So If I buy a module and build a pcb-board myself, can I already program and test it with arduino ?