GrumpyOldPizza/ArduinoCore-stm32l0 #192: Recieve transmissions for 1 day, nothing for 3 days, receive transmissions again.

kevin192291, 2021-07-02:
Hello,
I have a board and it is setup to transmit every 4 hours, then go to sleep. Everything was great for 1 day
then the board seemed to stop transmitting. I restart it, things go well again for another day and stop again.

I tried to leave it running and after 3 days of no transmissions I received a message just this morning.
I am wondering if it has anything to do with this: https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/27

Any insight would be greatly appreciated!

HeadBoffin, 2021-07-27:
Are you using ABP or OTAA?

kevin192291, 2021-07-28:
> Are you using ABP or OTAA?

I am using OTAA.

Here is my code with all of the boring stuff omitted:

`
void setup() {
  Serial.begin(9600);
 ...just a bunch of init stuff here
  
  LoRaWAN.begin(AS923);
  //LoRaWAN.setAntennaGain(3.0);
  LoRaWAN.joinOTAA(appEui, appKey, devEui);
  LoRaWAN.onReceive(receiveCallback);
  Serial.println("Leaving Setup");
}

void loop(void) {
  if (!LoRaWAN.busy())
  {
        if (CountDown)
        {
            CountDown--;
            
            if (!CountDown)
            {
                Serial.end();
                USBDevice.detach();
            }
        }

        if (!LoRaWAN.linkGateways())
        {
            if (CountDown)
            {
                Serial.println("REJOIN( )");
            }
            Serial.println("Rejoining...");
            LoRaWAN.rejoinOTAA();
        }
        
        if (LoRaWAN.joined())
        {
          Serial.println("LoRaWAN Joined");
            if (CountDown)
            {
              Serial.print("Count Down: ");
              Serial.println(CountDown);
            }

.... a bunch of LPP stuff here
            
            if (LoRaWAN.getMaxPayloadSize() > myLPP.getSize()) {
              LoRaWAN.sendPacket(myLPP.getBuffer(), myLPP.getSize());
              myLPP.reset();
              digitalWrite(powerPin, HIGH);
              digitalWrite(gpsEnPin, LOW);
              Serial.println("Going To Sleep USB will end now zzz...");
              STM32L0.stop(sleepTime);
            } else {
              Serial.println("Can't send too large of a packet in the start, lets just say hi!");
              LoRaWAN.beginPacket();
              LoRaWAN.write(0x68);
              LoRaWAN.write(0x69);
              LoRaWAN.endPacket();
              myLPP.reset();
              digitalWrite(powerPin, HIGH);
              digitalWrite(gpsEnPin, LOW);
              Serial.println("*NOT* Going To Sleep USB will end now zzz...");
            }
            
        }
    }
}
`