GrumpyOldPizza/ArduinoCore-stm32l0 #200: Connecting LoraWAN to TTNv3 error MIC mismatch

rojikada, 2022-04-11:
Hello, does anybody have this working with the TTNv3? I am trying to use the lorawan module with TTNv3, however in the TTN console I get MIC mismatch.
The appEui = appEUI, DevEUI=devEui, JoinEUI=appKey and the specifiation is LoRaWAN 1.0.2 ?

rojikada, 2022-04-11:
Aha! The problem was in having all zeroes in appEUI, after recreating the app with non-zero appEUI it started to work!

closed 2022-04-11 by rojikada

salmanfarisvp, 2022-10-18:
Hi @rojikada , where did you get the the appEUI , devEUI and JoinEUI -> Did you generate from your own? Thanks.

rojikada, 2022-10-18:
I just changed it within my code and the TTN web

salmanfarisvp, 2022-10-19:
I see the JOINEUI is not available from the TTN, but another key can be generated. So can we use a random for that and match the same with the node firmware?

rojikada, 2022-10-19:
I don’t understand what are you trying to do.

And as stated previously JoinEUI=appKey, which worked for me.

salmanfarisvp, 2022-10-19:
Sorry for the confusion, I mean: When we register a new node in TTN, we need to provide the JoinEUI, so for a DIY project where do we get the JoinEUI if the manufacture did provide it with all?  Thanks. 

[image]

rojikada, 2022-10-19:
You just generate one and put it to your code setup. If the device is not programmable, you might be on the wrong page.

salmanfarisvp, 2022-10-19:
Got it, Thanks for your time. 🙂

I use the http://descartes.co.uk/CreateEUIKey.html to generate the EUI key.