GrumpyOldPizza/ArduinoCore-stm32l0 #203: Setting spreading factor

SmolBumblebee, 2022-07-18:
Hi, is there a way for me to set the spreading factor? 

I've also been trying to set the data rate, but no matter what value I put it will always print 2 on the serial monitor, and my max payload size remains as 11. I am hoping to get 125 bytes for the max payload. 

Thank you

GrumpyOldPizza, 2022-07-18:
The spreading factor is set via setDataRate(). The data rate is region
specific. So you need to check out the LoRaWAN spec as to what setting you
want to use there.

SmolBumblebee, 2022-07-20:
Thank you so much for your fast reply; I feel so honored to get a response from GrumpyOldPizza ^_^

It turns out that I need to set ADR(false) for the setDataRate() to affect the max payload size. 

Is there documentation for the code by chance? 
I'm a new-bee