GrumpyOldPizza/ArduinoCore-stm32l0 #206: spi1

tcpipchip, 2022-11-22:
Hi
Is it working ?

GrumpyOldPizza, 2022-11-22:
He's. If the variant you are using supports it ... However I would think that none of the variants exposes a SPI1 for STM32L0.

tcpipchip, 2022-11-23:
yes, no support the variantes to STM32L0
I will have to add the SPI1, because i use the LOM204 module, a STM32L073 module with SX1276 in SPI...i will have to add a ENC28CJ60 on SPI1...
But...hummm....why  am not use the SPI to both SX1276 adn ENC28j60 ? Right ? The difference will be the NSS, right ?

Btw, i did a blog using RADIOHEAD...reference to your ARDUINO CORE STML0 lib :)
https://lom204-cli-wisol.blogspot.com/2022/11/lom204a-criando-uma-rede-mesh-com.html

GrumpyOldPizza, 2022-11-23:
I see. The way this code works with LoRaWAN is to basically bypass the Arduino layer and talk to SPI directly internally. There is code in ArduinoCore-stm32l0/tree/master/system/STM32L0xx/Source/LoRa/Boards that does that. 

I have not been able to find any docs for LOM204 at all ...

UT2UH, 2022-11-27:
@tcpipchip - can you share LOM204 docs? And the price/availability as well. I will add its support both to the core fork and RadioHead.

tcpipchip, 2022-11-27:
good morning!!!!!!!
[Datasheet_LOM204A_V0.1_20190715.pdf](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/files/10098310/Datasheet_LOM204A_V0.1_20190715.pdf)
You can get in SOUTH AMERICA
https://loja.smartcore.com.br/modem-lora-para-rede-publica-lorawan-wisol-lom204a-br00
Price, about $26,00 one piece

tcpipchip, 2022-11-27:
let me know if you need more info about LOM204...
[Wisol - carta para acompanhar WSLOM204A02 - SmartCore - p2p -20220901.docx](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/files/10098328/Wisol.-.carta.para.acompanhar.WSLOM204A02.-.SmartCore.-.p2p.-20220901.docx)

tcpipchip, 2022-11-27:
> @tcpipchip - can you share LOM204 docs? And the price/availability as well. I will add its support both to the core fork and RadioHead.

About RadioHead fork, can you make it compatible with STM32 ARDUINO GITHUB ?
https://github.com/stm32duino/Arduino_Core_STM32
Because we want to make compatible with https://github.com/khoih-prog STM32L0 examples :), where i was a collaborator...

UT2UH, 2022-11-27:
> let me know if you need more info about LOM204... 
Do you have MCU-SX1276 pinout?

UT2UH, 2022-11-27:
> About RadioHead fork, can you make it compatible with STM32 ARDUINO GITHUB ? 
Sorry, no. I use RadioHead but only with this core.

tcpipchip, 2022-11-27:
[GPIOS_LOM204_GIT.docx](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/files/10098400/GPIOS_LOM204_GIT.docx)

tcpipchip, 2022-11-27:
have you done some gateway with RADIOHEAD on STM32 ?
Wifi, ethernet, GSM, etc
Today i use only the UART

UT2UH, 2022-11-27:
> have you done some gateway with RADIOHEAD on STM32 ? Wifi, ethernet, GSM, etc 
Sorry, no.

tcpipchip, 2022-11-27:
OK